roscopetracula/hindenburgFights #19: Do motor faults cause the motor to shut down, and do we handle it correctly?

jef-pearlman, 2015-05-02:
This may already be handled in the existing code; I just want to read up on it and make sure we're handling it intelligently.  Different faults may also behave differently.

4levity, 2015-07-23:
since I am writing battery fault code I will examine this issue as well!

jef-pearlman, 2015-07-23:
Cool.  In particular, if the motors do shut down, I've been meaning to
check that we re-set the motor state after clearing the faults.  (I'm
actually pretty sure we don't, but depending on the answer on how it
behaves, we may or may not need to.)

We also don't really see faults in normal use--only when powered off USB,
or when the tether wraps around the motor and freezes it, or possibly when
low on batteries (which we should handle more intelligently soon anyway).
But still good to have it behave correctly.

4levity, 2015-08-13:
TL;DR: yes, handled in software

1. The current PCB does not have a resistor to enable ILIMIT current limit fault, the sense pin is connected to ground. Jef has seen this fault during testing, it may have been a glitch. If ILIMIT fault is triggered the controller does not disable the motor (but our software will, see below).

2. Motor controller will disable the FETs momentarily while the fault condition is in place. However our code resets the fault as soon as it is detected which would enable the FETs again. (but we also enter a fault-cooldown state, see below)

3. Testing indicated that temp/protection faults are NOT seen by just manually stalling the motor. 

4. Current blimp code WILL put the motor into coast and cool down for a bit if a fault is seen (although the fault on the controller itself gets reset).

closed 2015-08-13 by 4levity