roscopetracula/hindenburgFights #22: Look for and integrate i2c-controlled I/O pins.

jef-pearlman, 2015-05-10:


4levity, 2015-07-23:
working on support for new board version with i2c expander!

was going to compile separate firmware for old/new boards, controlled with #defines. but it seems like a better idea to look for i2c expander at runtime and automatically configure.

jef-pearlman, 2015-07-23:
May I suggest reporting back to the client (a string probably) if no expander is found?  Eventually we shouldn't have boards without them, and it'll let us know if anything malfunctions.

4levity, 2015-07-24:
i2c expander detection, automatic pin assignment, and use of LEDs is implemented but not tested on new board yet (mostly tested to not break old board)
 
remaining: 
* get it actually working with expander
* notify controller (fault?) if i2c expander NOT found

4levity, 2015-08-13:
All tested and working now.

closed 2015-08-13 by 4levity